pymmcore-plus/pymmcore-widgets #101: task: remove metadata about grids from position name

tlambert03, 2022-12-06:
I noticed in https://github.com/pymmcore-plus/napari-micromanager/pull/234/ that the metadata needed to assign different positions in an explorer experiment to their respective locations in the grid is stored in the `position.name` as a string like "Grid_001_Pos000", and then parsed by napari-micromanager by splitting on underscores to recover the metadata.

This should be changed to not use string parsing, but rather use `sequence.metadata`.  For example in this line: https://github.com/pymmcore-plus/pymmcore-widgets/blob/aeb59bc4de5823cdd5fd579405952b6e551f681a/src/pymmcore_widgets/_sample_explorer_widget/_sample_explorer_widget.py#L461-L466

rather than passing stage_positions with names holding that info, we could use `sequence.metata["grid_info"] = {(x,y,z): (grid, grid_pos)}` storing a mapping of x,y,z location to additional metadata about that position.  (it even kinda seems like x,y,z is _already_ sufficient to know the row column info? ... and membership in any specific grid could go in the metadata).

Once done, the logic in MDAHandler should be updated accordingly.

https://github.com/pymmcore-plus/napari-micromanager/blob/b6922ab8c9c7523b212557f3b46bcb83f9c50f2c/src/napari_micromanager/_mda_handler.py#L239

tlambert03, 2023-07-25:
@fdrgsp, can you update the status here, and close if necessary?

fdrgsp, 2023-07-25:
closing since explorer widget has been removed.

closed 2023-07-25 by fdrgsp