pymmcore-plus/pymmcore-widgets #156: Mda widget broken

ianhi, 2023-07-26:
* pymmcore-widgets version: main
* Python version:
* Operating System: linux

### Description

Run the `example/mda_widget.py`
get this error:

```python
Traceback (most recent call last):
  File "/home/ian/Documents/oss/micro/pymmcore-widgets/examples/mda_widget.py", line 99, in <module>
    frame = MDA()
  File "/home/ian/Documents/oss/micro/pymmcore-widgets/examples/mda_widget.py", line 49, in __init__
    self.mda = MDAWidget(include_run_button=True)
  File "/home/ian/Documents/oss/micro/pymmcore-widgets/src/pymmcore_widgets/_mda/_mda_widget.py", line 211, in __init__
    self._on_sys_cfg_loaded()
  File "/home/ian/Documents/oss/micro/pymmcore-widgets/src/pymmcore_widgets/_mda/_mda_widget.py", line 218, in _on_sys_cfg_loaded
    self._enable_run_btn()
  File "/home/ian/Documents/oss/micro/pymmcore-widgets/src/pymmcore_widgets/_mda/_mda_widget.py", line 268, in _enable_run_btn
    if self.ch_cbox.isChecked() and not self.channel_widget._table.rowCount():
AttributeError: 'QWidget' object has no attribute 'isChecked'
```

attn @fdrgsp

fdrgsp, 2023-07-26:
@ianhi which qt are you using?

ianhi, 2023-07-26:
Looks like I'm using QT5 bindings version 5.15.7

ianhi, 2023-07-26:
hmm, in a new environment it's not replicatingthe same. Maybe somethign weird there. closing for now

closed 2023-07-26 by ianhi