pymmcore-plus/pymmcore-widgets #167: MDAWidget should be able to load/create MDASequence without core

tlambert03, 2023-07-29:
I *think* this should be possible, but let me know @fdrgsp if you disagree.

It would be nice if all of the widgets that input/output generic schema objects from useq-schema could do that without having a system configuration loaded in core.  For example, it seems like you should be able to use `MDAWidget.get_state()` to create an MDASequence object even without the core being loaded (imagine a user not on the microscope creating an experiment and sending a json or yaml file to some other computer).  Similarly, it should be able to load the state of a Sequence without having a full core attached, but currently, you get something like this:

```pytb
In [4]: wdg = MDAWidget()

In [5]: wdg.set_state(mda)
/Users/talley/dev/self/pymmcore-widgets/src/pymmcore_widgets/_mda/_mda_widget.py:314: UserWarning: 'DAPI' config or its group doesn't exist in the 'Channel' ChannelGroup!
  self.channel_widget.set_state([c.dict() for c in state.channels])
/Users/talley/dev/self/pymmcore-widgets/src/pymmcore_widgets/_mda/_mda_widget.py:314: UserWarning: 'FITC' config or its group doesn't exist in the 'Channel' ChannelGroup!
  self.channel_widget.set_state([c.dict() for c in state.channels])
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[5], line 1
----> 1 wdg.set_state(mda)

File ~/dev/self/pymmcore-widgets/src/pymmcore_widgets/_mda/_mda_widget.py:335, in MDAWidget.set_state(self, state)
    333 if state.stage_positions:
    334     self.p_cbox.setChecked(True)
--> 335     self.position_widget.set_state(list(state.stage_positions))
    336 else:
    337     self.p_cbox.setChecked(False)

File ~/dev/self/pymmcore-widgets/src/pymmcore_widgets/_mda/_positions_table_widget.py:723, in PositionTable.set_state(self, positions, clear)
    720     raise TypeError("The 'positions' arguments has to be a 'Sequence' type.")
    722 if not self._mmc.getXYStageDevice() and not self._mmc.getFocusDevice():
--> 723     raise ValueError("No XY and Z Stage devices loaded.")
    725 # variables used for autofocus devaice name check
    726 rows = set(range(self._table.rowCount()))

ValueError: No XY and Z Stage devices loaded.
```

fdrgsp, 2023-07-29:
Correct! I agree. I will open a PR to change this behavior.

And actually just by removing  this line from the `set_state()` method of the `PositionTable` fixes the issue. 

```py
if not self._mmc.getXYStageDevice() and not self._mmc.getFocusDevice():
     raise ValueError("No XY and Z Stage devices loaded.")
```
After removing the line above, running the code below works as expected.
```py
from useq import MDASequence
from pymmcore_widgets import MDAWidget
from pymmcore_plus import CMMCorePlus
mmc = CMMCorePlus.instance()
mmc.loadSystemConfiguration()
mmc.unloadDevice("XY")
m = MDAWidget()
m.show()
mda = MDASequence(stage_positions=[(1,2,3), (4,5,6)])
m.set_state(mda)
```

fdrgsp, 2023-07-29:
> For example, it seems like you should be able to use MDAWidget.get_state() to create an MDASequence object even without the core being loaded

But if the core is not involved, how can you get the channel information or the coordinates of a stage for example?

tlambert03, 2023-07-29:
yep, good point.  so it's clear that the ability to _populate_ an MDASequence with specific information about a channel or a current position will definitely depend on having an active microscope.  but I think it's safe to say that we should be able to load anything without a core.  So perhaps just the line you mentioned above is all we need?

fdrgsp, 2023-07-29:
I opened a PR https://github.com/pymmcore-plus/pymmcore-widgets/pull/169.

closed 2023-07-31 by tlambert03