pymmcore-plus/pymmcore-widgets #240: Pass step size to a StageWidget?

dpshepherd, 2023-10-31:
When creating a new StageWidget for a custom GUI, is there a way to pass in a step size to override the default? We have a piezo stage that I would like to set the default to something smaller than 10 (0.1 specifically). I looked at the docs and the function docstring, but didn't see anything immediately.

Or maybe the preferred way to do this to create the Widget then set the step size? If so, can you point me in the right direction to do this when using just a StageWidget?

Thanks!

tlambert03, 2023-10-31:
looks like that hasn't been publicly exposed, but absolutely should.  I'll add it now

tlambert03, 2023-11-01:
v0.5.6 contains the fix and is building and deploying to PyPI currently.  should be up in ~15 mins or so.  cheers!

closed 2023-11-01 by tlambert03

dpshepherd, 2023-11-01:
Wow, instant. Shouldn't you both be out getting some candy? 🤣

Thank you!!

PR #241 (2023-11-01): feat: add step() and setStep() methods to StageWidget
closes #240 

@dpshepherd, after this, it would be:

```python
wdg = StageWidget('deviceLabel', step=1)
# or later
wdg.setStep(...)
```