pymmcore-plus/pymmcore-widgets #256: MDA interval input behavior

wl-stepp, 2023-12-19:
* pymmcore-widgets version: 0.1.dev190+g3f1e2ea
* Python version: 3.11.4
* Operating System: Windows 10

### Description

The interval input is behaving a little unintuitively in my opinion. See video attached

### What I Did

https://github.com/pymmcore-plus/pymmcore-widgets/assets/52414717/0c0a0bf0-8b38-45bc-b108-ac0eda0b0e87

tlambert03, 2023-12-19:
Can you point out what exactly is the unintuitive bit?  Is it that 60 loops with a 1 second interval would have a duration of 59 seconds?

wl-stepp, 2023-12-19:
When I mark everything and type 0.5 I get 5us, when I type 1 it's 1s.

tlambert03, 2023-12-19:
ahh... ok, thanks for clarifying.  yeah i agree, that's incorrect.  let me see what's going on with the units

tlambert03, 2023-12-19:
fix in v0.5.7, coming shortly

PR #257 (2023-12-19): fix: fix quantity edit
fixes #256
closed 2023-12-19 by tlambert03