pymmcore-plus/pymmcore-widgets #300: GroupPresetTableWidget preset addition causes error with signal block

wl-stepp, 2024-06-12:
Package                       Version
----------------------------- -----------
annotated-types               0.7.0
appdirs                       1.4.4
click                         8.1.7
colorama                      0.4.6
flexcache                     0.3
flexparser                    0.3.1
fonticon-materialdesignicons6 6.9.96
freetype-py                   2.4.0
hsluv                         5.0.4
kiwisolver                    1.4.5
markdown-it-py                3.0.0
mdurl                         0.1.2
ml-dtypes                     0.4.0
numpy                         1.26.4
packaging                     24.1
Pint                          0.24
pip                           24.0
platformdirs                  4.2.2
psygnal                       0.11.1
pydantic                      2.7.3
pydantic-compat               0.1.2
pydantic_core                 2.18.4
Pygments                      2.18.0
pymmcore                      11.1.1.71.0
pymmcore-plus                 0.10.1
pymmcore-widgets              0.7.1
PyQt5                         5.15.10
PyQt5-Qt5                     5.15.2
PyQt5-sip                     12.13.0
QtPy                          2.4.1
rich                          13.7.1
setuptools                    65.5.0
shellingham                   1.5.4
superqt                       0.6.7
tensorstore                   0.1.61
tifffile                      2024.5.22
typer                         0.12.3
typing_extensions             4.12.2
useq-schema                   0.4.7
vispy                         0.14.2
wrapt                         1.16.0

Windows 10 Enterprise

### Description

When adding a preset group I get get this error

WARNING: Traceback (most recent call last):
  File "D:\python_envs\eda_env\Lib\site-packages\pymmcore_widgets\_group_preset_widget\_add_first_preset_widget.py", line 148, in _create_first_preset
    with block_core(self._mmc.events):
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\python_envs\eda_env\Lib\site-packages\pymmcore_widgets\_util.py", line 93, in block_core
    return mmcore_events.blocked()  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'CMMCoreSignaler' object has no attribute 'blocked'

### What I Did

Add group in GroupPresetTableWidget

tlambert03, 2024-06-12:
@fdrgsp, can you work on this one?  Looks to be this function, added in #25 

https://github.com/pymmcore-plus/pymmcore-widgets/blob/5b76828cd716a3ef1d2c3ba9b93a9159e31e5b2e/src/pymmcore_widgets/_util.py#L90-L95

there's a type: ignore there covering up an actual issue that @wl-stepp is hitting here

tlambert03, 2024-06-13:
quick note @wl-stepp, we'll have this fixed later to day, but in the meantime, as a hacky workaround, I suspect you won't actually hit this error if you instantiate the `QApplication([])` *before* creating a `CMMCore` object.  (don't feel like you have to do that though.)

PR #302 (2024-06-13): fix: fix blocker
fixes #300, requires https://github.com/pymmcore-plus/pymmcore-plus/pull/357
closed 2024-06-13 by tlambert03