pymmcore-plus/pymmcore-widgets #301: StageWidget freezes application

wl-stepp, 2024-06-12:
Hello everyone,

we are installing pymmcore-plus and widgets on a Nikon Spinning Disk. If we move the stage from the StageWidget, or even only if the stage widget is loaded and we move the stage programmatically, the application freezes. I think the problem is in the callback linked below. If I comment that callback out everything seems to be working fine. Same if we are in MMStudio, no problems there with the same config. Let me know if I can provide further info.

pymmcore                      11.1.1.71.0
pymmcore-plus                 0.10.1
pymmcore-widgets              0.7.1

https://github.com/pymmcore-plus/pymmcore-widgets/blob/5b76828cd716a3ef1d2c3ba9b93a9159e31e5b2e/src/pymmcore_widgets/_stage_widget.py#L303

Device,XYStage,NikonTi2,XYStage

tlambert03, 2024-06-12:
hey @wl-stepp, i'm unable to reproduce this one. two way communication works fine, widget updates the stage, and stage movement with the joystick updates the widget.

so let's dig deeper into versions.  here is the output of my `mmcore info`:

```bash
(mmgui) C:\Users\NIC\Desktop\mm\pymmcore-widgets>mmcore info
python            : 3.11.8 | packaged by Anaconda, Inc. | (main, Feb 26 2024, 21:34:05) [MSC v.1916 64 bit (AMD64)]
platform          : Windows-10-10.0.19045-SP0
pymmcore-plus     : 0.10.1
pymmcore          : 11.1.1.71.0
core-version-info : MMCore version 11.1.1
api-version-info  : Device API version 71, Module API version 10
adapter-path      : ~\AppData\Local\pymmcore-plus\pymmcore-plus\mm\Micro-Manager_2.0.3_20231128
useq-schema       : 0.4.7
pymmcore-widgets  : 0.7.2.dev0+gee0fcb47.d20240612
qt                : PyQt6 6.6.1
tifffile          : 2024.2.12
zarr              : 2.17.1
```

and the file version of my Ti2_Mic_Driver.dll is `2.2.0.169`

wl-stepp, 2024-06-12:
Newer device adapter does not seem to help. We are using newest nightly 20240611
Is this maybe hardware specific, might the stage not like a call to getXYPosition during movement or something like that?

closed 2024-06-19 by wl-stepp

tlambert03, 2024-06-19:
learn anything?

wl-stepp, 2024-06-19:
No, will have a look next time I'm on that specific scope and reopen if I find something.

wl-stepp, 2024-07-11:
We had to reinstall the environment due to different reasons and there seems to be no issue with these versions.
Maybe it was related to #300 ?

pymmcore                      11.1.1.71.1
pymmcore-plus                 0.10.2
pymmcore-widgets              0.7.2

tlambert03, 2024-07-11:
good!  

> Maybe it was related to https://github.com/pymmcore-plus/pymmcore-widgets/issues/300 ?

maybe!