pymmcore-plus/pymmcore-widgets #312: TODO: remove grid iteration in MoveToWidget._move_to_row_col

tlambert03, 2024-07-05:
@fdrgsp, here's another one for ya:

I think these lines here are the only place in pymmcore-widgets that you directly access the `.row` or `.col` of a GridPosition object, which I'm trying to remove in https://github.com/pymmcore-plus/useq-schema/pull/169

https://github.com/pymmcore-plus/pymmcore-widgets/blob/1d19b1a10f963294b38cd3bcb969f25c49d0054e/src/pymmcore_widgets/_mda/_grid_widget.py#L553-L554

it also happens to be an unnecessary usage of `row/column` and `iter_grid_positions` in the first place :) and there's a note there to remove it.  Basically, it looks like in order to move the stage to row X, col Y, you exhaustively iterate through the positions of the entire plate looking for the one that matches the row/col that you want to move to.  It seems to me that all you need to know is the fov height/width and overlap, and a base offset?  Could you try to remove that block?

fdrgsp, 2024-07-05:
Hey @tlambert03, this is from the old version of the widgets, aren't we going to get rid of them at some point?

tlambert03, 2024-07-06:
ah, yeah, I wondered why I couldn't find it :)  
can I just delete it then along with the old MDA widgets?

fdrgsp, 2024-07-06:
> ah, yeah, I wondered why I couldn't find it :) can I just delete it then along with the old MDA widgets?

yes I think so!

PR #313 (2024-07-06): refactor: remove old MDA widget
closes #312
closed 2024-07-06 by tlambert03