pymmcore-plus/pymmcore-widgets #324: Allow Config Groups to be Renamed

gselzer, 2024-07-12:
* pymmcore-widgets version: 0.7.2
* Python version: 3.12.4
* Operating System: Windows 11

### Description

Is there a reason why we cannot rename config groups by clicking the `Edit` button with a group selected?

### What I Did

With the demo config active, I clicked the `Camera` group, and then clicked the `Edit` button. There's a `QLineEdit` for the group name, but it cannot be edited.

[image]

fdrgsp, 2024-07-13:
Thanks @gselzer! This is actually a bug. I just fixed it [here](https://github.com/fdrgsp/pymmcore-widgets/tree/fix-edit-group) so if you want to try it out you can pip install this branch: 
```
pip install git+https://github.com/fdrgsp/pymmcore-widgets@fix-edit-group
```

tlambert03, 2024-07-13:
@fdrgsp, if you have a proposed fix, it's best to open a PR, and then people can very easily `gh co #NUMBER`

fdrgsp, 2024-07-13:
👍🏻 here's the PR #328

PR #328 (2024-07-13): fix: make name editable EditGroupWidget
This PR makes the group name editable when interacting with the edit group dialog in the `GroupPresetTableWidget`.

closes #324
closed 2024-07-13 by tlambert03