pymmcore-plus/pymmcore-widgets #326: Add an Autofocus widget

gselzer, 2024-07-12:
If there are loaded `AutoFocusDevice`s, it'd be nice to have a widget for running them!

marktsuchida, 2024-07-16:
+1 but I would suggest that "hardware focus-maintenance devices" (PFS/AFC/DeFo/ZDC/others) should be treated separately from "software autofocus" (which should probably be implemented on top of MDAEngine). Conflating the two was probably a mistake in MMStudio. (It is true that there are scenarios in which they are used in a similar manner, so there may be cases where they happen show up in the same part of the GUI, but that is only part of the story.)

(I think @gselzer is thinking of software AF, given where this is coming from, which might need to first exist in pymmcore-plus (or somewhere) before a widget can be created.)

tlambert03, 2024-07-16:
ah, yeah i assumed we were talking about hardware focus, in which the widget would basically just be a status indicator (locked or not) and probably a stage offset or something?

marktsuchida, 2024-07-16:
I think that would be quite useful, too (though our lab doesn't have any HW focus devices...).