pymmcore-plus/pymmcore-widgets #335: MDA Save button is ambiguous

gselzer, 2024-07-16:
* pymmcore-widgets version: 0.7.2
* Python version: 3.12.4
* Operating System: Windows 11

### Description

A colleague noted that the `Save` button on the `MDAWidget` is not particularly clear - he didn't know if "Save" pertained to the MDA parameters or to the acquired images. Could we think of some addition (either changing the button text or adding a label) that could make it clearer that the parameters can be saved/loaded?

tlambert03, 2024-07-16:
Sure, "Save Settings?"  "Save MDA?"  "Save parameters?"

gselzer, 2024-07-16:
Hmm - I think either of "Save Settings", `Save Param(eter)s" would be great!

tlambert03, 2024-07-16:
care to join us as a contributor? :)

gselzer, 2024-07-16:
> care to join us as a contributor? :) 

Absolutely, once I finish converting this long list of action items (which includes this point) into issues I'd love to contribute!

tlambert03, 2024-07-16:
ok 😄 ... can I leave this particular issue for you then?

PR #346 (2024-07-18): style: clarify save/load buttons in MDAWidget
I believe that these alterations make it clearer that the buttons save and load MDA parameters, and not images/devices/etc.

I do wonder if the text is a little wide though, what do others think?

[image]

Closes #335
closed 2024-07-18 by tlambert03