pymmcore-plus/pymmcore-widgets #336: Make it clear that GroupPresetTableWidget slider values can be edited

gselzer, 2024-07-16:
* pymmcore-widgets version: 0.7.2
* Python version: 3.12.4
* Operating System: Windows 11

### Description

As I alluded to in [this comment](https://github.com/pymmcore-plus/pymmcore-widgets/pull/334#discussion_r1679930465), my colleague did not notice that the `QDoubleSpinBox` in the `GroupPresetTableWidget` could be edited. It'd be great to provide some visual clue (e.g. spin buttons) to indicate that it's editable, and not just a label.

[image]

tlambert03, 2024-07-16:
that's a widget from superqt.  
I'm a little bit conflicted on that one actually.  Those are ugly and take up valuable space; and once you notice they're editable, you know it forever.  
I can show you how to enable them via private methods (and we might want to add a public feature for that at superqt); I can see that someone might want it as an option;  but I'm not immediately sure I want to make it the default appearance for everyone.

gselzer, 2024-07-17:
> once you notice they're editable, you know it forever.

That's a fair point. There are a couple other things I can think of that could be helpful:
* Add a tooltip to the SpinBox that somehow implies editability
* Edit the border style of the SpinBox - that could be done within my UI, no change needed here, right?

marktsuchida, 2024-07-17:
I was thinking the same things. A tooltip on the combined slider-box widget could indicate which property it controls (as with the regular config groups), plus mention "click the number to enter a value".
Also, we should probably watch more people use it before concluding that we absolutely need to emphasize editability here.

tlambert03, 2024-07-17:
yeah, i think this is something that the app-builder should do with style sheets.  superqt does currently setStyleSheet on the slider label, but it shouldn't... and after https://github.com/pyapp-kit/superqt/pull/254, you could do this yourself with a stylesheet:

```python
from qtpy.QtWidgets import QApplication
from superqt import QLabeledSlider

app = QApplication([])
app.setStyleSheet(
    "SliderLabel { border: 2px solid black; background-color: yellow; }"
)
sld = QLabeledSlider()
sld.show()
app.exec()
```

[image]