pymmcore-plus/pymmcore-widgets #337: GridPlanWidget buttons set incorrect bounds

gselzer, 2024-07-16:
* pymmcore-widgets version: 0.7.2
* Python version: 3.12.4
* Operating System: Windows 11

### Description
Using the Stage Control, we tried to set up a Grid Plan using the `MDAWidget`. When we clicked on the corner buttons, we found:
| Button | Bound 1 | Bound 2 |
|--------|----------|----------|
| top left | top | right |
| top right | top | bottom |
| bottom left | left | right |
| bottom right | left | bottom |
| left | right | | 
| right |  bottom | |
| top | top | |
| bottom | left | | 

### What I Did

1. Started up [my UI](https://github.com/gselzer/pymmcore-plus-sandbox)
2. Opened up `Stage Control` (A combination of `StageWidget`s) and `MDA` (an `MDAWidget`)
3. Moved to `(X, Y)=(20, 20)
4. Clicked the Top Left button

This resulted in the state below:

[image]

gselzer, 2024-07-16:
> Hi @gselzer! I think this is because we are using a cartesian coordinate system where top left is (-x +y) and bottom right is (+x -y). Therefore you need to transpose your stage axis in the micromanager properties of the stage (e.g. you can open the PropertyBrowser widget and set it from there). Does it make sense?
> 
> And then to match the movement in the stage widget, just toggle the Invert X and/or Invert Y checkboxes.

Hi @fdrgsp! I tried toggling `XYTransposeMirror-X`, but this did not change things. Furthermore, I do not believe that any toggling could resolve that the Top Right button sets both the top and bottom border. Am I missing something?

tlambert03, 2024-07-16:
> Therefore you need to transpose your stage axis in the micromanager properties of the stage (e.g. you can open the PropertyBrowser widget and set it from there). Does it make sense?

shouldn't we be able to deal with whatever the user enters?  That is: regardless of how they've set up the microscope, if they mark top/bottom/left/right in *any* way, shouldn't that be enough for us to recreate the bounds that they want?

gselzer, 2024-07-17:
> shouldn't we be able to deal with whatever the user enters? That is: regardless of how they've set up the microscope, if they mark top/bottom/left/right in _any_ way, shouldn't that be enough for us to recreate the bounds that they want?

I'd think so - you can feel free to assign this issue to me!

PR #345 (2024-07-18): fix: Align spin boxes and labels in GridPlan
I believe that the behavior I described in #337 was a simple misordering of the `_PositionSpinBox`es. I reordered them and things seem to work as described!

Closes #337
closed 2024-07-18 by tlambert03