pymmcore-plus/pymmcore-widgets #339: All Radio Buttons appear On within GridPlan widget

gselzer, 2024-07-17:
* pymmcore-widgets version: 0.7.2
* Python version: 3.12.4
* Operating System: Windows 11

### Description
A colleague was confused by the radio buttons within the `GridPlanWidget`, as they appeared to him to all be on at once - it's not apparent at a glance which is actually on.

[image]

The UI does not have this problem on macOS (at least as far as I can tell on the documentation screenshots) nor on WSL, so it might be a OS-specific thing we can't do much about, or maybe it is a styling thing?

tlambert03, 2024-07-17:
this is what I see on windows 10:

[image]

what does yours like like without the dark theme?

gselzer, 2024-07-17:
Nope 😅

[image]

Windows 11 thing, then?

tlambert03, 2024-07-17:
ok,  yeah, i think it's the accent color on windows 11 somehow...  try playing with your personalization stuff and see if you can learn anything.  That circle in the first one *is* showing that it's selected...

[image]

but it's obviously incredibly hard to see.  windows is annoying.  I also don't know why your other fields don't seem to be disabled.   this is the standard example right?  Not a widget within a broader app that might have a style sheet applied?

gselzer, 2024-07-17:
> try playing with your personalization stuff and see if you can learn anything. 

Yeah, I'll keep looking.

> windows is annoying. 

💯

> I also don't know why your other fields don't seem to be disabled like they are in my view. 

Well, they *are* disabled, but there's no visual indication (outside of the cursor not changing to the I-Beam on the disabled fields) ☹️ 

> this is the standard example right? Not a widget within a broader app that might have a style sheet applied?

The above image was from my UI, however it looks identical if I run `python pymmcore-widgets/examples/mda_widget.py`

gselzer, 2024-07-18:
So this MCVE showcases the problem. On Windows 11, it would seem that the line `self.setWidget(self.inner)`  is the culprit that affects the style, as removing that line removes the accent fill that makes this so confusing.

```python
from qtpy.QtWidgets import QApplication, QScrollArea, QWidget, QRadioButton, QVBoxLayout

app = QApplication([])

class FooWidget(QScrollArea):

    def __init__(self, parent: QWidget | None = None):
        super().__init__(parent)

        self.btn = QRadioButton()
        
        self.inner = QWidget(self)
        layout = QVBoxLayout(self.inner)

        layout.addWidget(self.btn)

        self.setWidget(self.inner)

wdg = FooWidget()
wdg.show()
app.exec()
```

gselzer, 2024-07-18:
Despite my [multiple](https://github.com/pymmcore-plus/pymmcore-widgets/pull/344#issue-2416539215) [attempts](https://github.com/pymmcore-plus/pymmcore-widgets/pull/344#issuecomment-2237165138) this issue was not automatically closed with the merge of #334 - so I'll do it myself! 😤

closed 2024-07-18 by gselzer