pymmcore-plus/pymmcore-widgets #342: GridPlanWidget's Overlap, Order, Relative To confusions

gselzer, 2024-07-17:
* pymmcore-widgets version: 0.7.2
* Python version: 3.12.4
* Operating System: Windows 11

### Description
A colleague was unaware that the `GridPlanWidget` options `Overlap`, `Order`, and `Relative To` pertained to all of the different mechanisms for specifying a `Grid Plan`. He did not even notice the `Relative To` dropdown, because it was hidden:

[image]

One fix could be ensuring that the widget is tall enough such that all tabs fit without the need for a scrollbar. Beyond that, maybe we could organize the layout differently such that it's clearer these options pertain to all methods of grid specification? How might we better do that? We could copy these options into each section of the `GridPlanWidget`, or add a box around the three radio button options to make it clearer that the options below are not an "additional choice"?

tlambert03, 2024-07-17:
could combine the top three into a QTabWidget, that would have the advantage of only showing the active settings, while always showing the general options below.   The downside is that tabs really don't imply "selection".  
Alternatively, could use a QStackedWidget, with selection buttons at the top determining which of the widgets is shown/active.  I think that might be done in the Z tab (which itself needs some work).

there's definitely many ways this widget could be improved.  Someone just needs to play around with it a bit

> We could copy these options into each section of the GridPlanWidget

that would be too much space i think

> or add a box around the three radio button options to make it clearer that the options below are not an "additional choice"?

open to that as well

> ensuring that the widget is tall enough such that all tabs fit without the need for a scrollbar

that is too screen/user specific to be a good solution.  for anything more than a couple hundred pixels, you pretty much always have to allow for the possibility that it won't fit wherever it's being placed

gselzer, 2024-07-23:
This is solved now with #351

closed 2024-07-23 by gselzer