pymmcore-plus/pymmcore-widgets #363: Device name with "-" crashes GroupPresetWidget

wl-stepp, 2024-10-04:
* pymmcore-widgets version: 0.1.dev248+g53e9ddf.d20240719
* Python version: 3.11.9
* Operating System: Windows 10

### Description

Trying to update a PresetConfig from the Widget by added another Preset to Channel. Crashes because one of our devices is called "pE-800" and the "-" makes the widget think it's called "pE" only. This works in the Java version however. I think there might be a step where the "-" is used to separate device and setting?

### What I Did
We can replicate with a minimal system:
```
from pymmcore_widgets import GroupPresetTableWidget
from pymmcore_plus import CMMCorePlus
from qtpy.QtWidgets import QApplication

app = QApplication([])
mmcore = CMMCorePlus()
mmcore.loadSystemConfiguration('./MMConfig_error.cfg')

group_presets = GroupPresetTableWidget(mmcore=mmcore)
group_presets.show()

app.exec_()
```
Error when trying to change a preset in ErrorChannel, config attached.
```
Traceback (most recent call last):
  File "C:\Control_2\pymmcore-widgets\src\pymmcore_widgets\_group_preset_widget\_group_preset_table_widget.py", line 227, in _on_new_group_preset
    self._populate_table()
  File "C:\Control_2\pymmcore-widgets\src\pymmcore_widgets\_group_preset_widget\_group_preset_table_widget.py", line 246, in _populate_table
    wdg = self._create_group_widget(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Control_2\pymmcore-widgets\src\pymmcore_widgets\_group_preset_widget\_group_preset_table_widget.py", line 281, in _create_group_widget
    return PropertyWidget(device, prop)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Control_2\pymmcore-widgets\src\pymmcore_widgets\_property_widget.py", line 312, in __init__
    raise ValueError(f"Device not loaded: {device_label!r}")
ValueError: Device not loaded: 'DCam'
```
[MMConfig_error.zip](https://github.com/user-attachments/files/17256393/MMConfig_error.zip)

tlambert03, 2024-10-04:
good catch and thanks for the issue.  this should be easy to fix

tlambert03, 2024-10-04:
this is fixed now.  side note: there's a lot to be desired in the ConfigGroup editing widgets.  The current ones are mostly a direct porting of the MMStudio interface (and they were made early in the process and contain a lot of code duplication).  So some variant of #307 or #338 will likely replace these widgets in the future.

PR #365 (2024-10-04): fix: fix splitting logic and deduplicate code in Groups Presets Widgets
fixes #363

This is just a patch for the immediate issue, these widgets still contain a lot of duplicated code and possible bugs, and should be replaced (with one of #307, #320, #338, etc...)
closed 2024-10-04 by tlambert03