pymmcore-plus/pymmcore-widgets #370: Add a Console Widget

gselzer, 2024-10-10:
It would be nice if we had a centralized console widget. For now, the main thing that would be nice is access to the `CMMCore(Plus)` instance, although we should certainly also have the API to add additional variables.

There are already a couple prototypes we could draw from. I added a `RichJupyterWidget`-based console to my sandbox [here](https://github.com/gselzer/pymmcore-plus-sandbox/blob/53ac7e8ca3b4874816583b8b74024a75432b8fc9/src/pymmcore_plus_sandbox/_console_widget.py#L5), but this may be overkill. @simonecoppola has additionally filed https://github.com/fdrgsp/micromanager-gui/pull/22, but that would require additional effort and bugfixes.

Thoughts @tlambert03 @fdrgsp @marktsuchida?

fdrgsp, 2024-10-11:
I agree, @gselzer. After @simonecoppola mentioned it in https://github.com/fdrgsp/micromanager-gui/pull/22, I also thought that `pymmcore-widgets` would be a good place for it.

tlambert03, 2024-10-11:
naive question (having not looked at anyone's code).  In a sense, isn't [qtconsole](https://qtconsole.readthedocs.io/en/stable/) already this widget?  And then, inasmuch as pymmcore-widgets is a component library (and not the final composed gui), what additional "wrapping" do we need to provide?  Given that it brings in enough [extra dependencies](https://github.com/jupyter/qtconsole/blob/a72387f01dea2076346ecb38857de5266414dbe6/setup.py#L68-L74) to make it a somewhat heavy addition, I think it should likely only be done by the final gui application (like @fdrgsp's app or https://github.com/pymmcore-plus/pymmcore-gui/issues).  in napari, we actually eventually extracted our console into an [independent package](https://github.com/napari/napari-console) for exactly this reason.

gselzer, 2024-10-11:
Yup, for the reasons you mentioned I nearly filed this issue within pymmcore-gui, but I guess the reason I filed it here is because at the end of the day we're talking about adding a new widget, and all the widgets (currently) live here. 

My main concern is avoiding duplicated effort, since we've already seen the beginnings of that. I think that the solution could occur here or on the application level, and see no problems with moving this issue over to pymmcore-gui.

tlambert03, 2024-10-11:
let's leave this here for the moment.  but if it gets added, I would only want it as an extra `pip install pymmcore-widgets[console]`

gselzer, 2025-03-12:
Since pymmcore-plus/pymmcore-gui#34 has made its way into release, I'm content to close this as unplanned for now. We can always reopen if we find we're duplicating that console logic across many different UIs.

closed 2025-03-12 by gselzer