pymmcore-plus/pymmcore-widgets #380: Bug: MDASequence useq widget does not load "include Z" option

wl-stepp, 2024-11-06:
* pymmcore-widgets version: pymmcore-widgets              0.1.dev248+g53e9ddf
* Python version: 3.11.9
* Operating System: Win10

### Description

When saving and reloading a MDASequence, the option to ignore Z will not be restored. Instead Z will always be there.

### What I Did

Slightly adjusted test_mda_wdg_load_save
```
import useq
from pymmcore_widgets.useq_widgets._mda_sequence import QFileDialog
from pymmcore_widgets.useq_widgets import MDASequenceWidget
import pymmcore_widgets
from qtpy.QtWidgets import QApplication
from pathlib import Path

MDA_noZ = useq.MDASequence(
    time_plan=useq.TIntervalLoops(interval=4, loops=3),
    stage_positions=[(0, 1), useq.Position(x=42, y=0)],
    channels=[{"config": "DAPI", "exposure": 42}],
    z_plan=useq.ZRangeAround(range=10, step=0.3),
    grid_plan=useq.GridRowsColumns(rows=10, columns=3),
    axis_order="tpgzc",
    keep_shutter_open_across=("z",),
)
ext= "json"
PYMMCW_METADATA_KEY = "pymmcore_widgets"

app = QApplication([])

wdg = MDASequenceWidget()
wdg.show()
import os
dest = Path(os.path.expanduser(f"~\Desktop\sequence.{ext}"))

dest.write_text("")

dest.write_text(MDA_noZ.yaml() if ext == "yaml" else MDA_noZ.model_dump_json())

wdg.load()
print(wdg.value().replace(metadata={}))
assert wdg.value().replace(metadata={}) == MDA_noZ

wdg.save()
mda_no_meta = MDA_noZ.replace(
    metadata={PYMMCW_METADATA_KEY: {"version": pymmcore_widgets.__version__}}
)

assert dest.read_text() == mda_no_meta.model_dump_json(exclude_defaults=True)

app._exec()
```

wl-stepp, 2024-11-06:
```
PS C:\Control_2> & c:/Control_2/.venv/Scripts/python.exe c:/Control_2/Zeiss-microscope/control_2/src/zeiss_control/backend/test_useq_save.py
axis_order=('t', 'p', 'g', 'z', 'c') stage_positions=(AbsolutePosition(x=0.0, y=1.0, z=0.0), AbsolutePosition(x=42.0, y=0.0, z=0.0)) grid_plan=GridRowsColumns(rows=10, columns=3) channels=(Channel(config='DAPI', exposure=42.0),) time_plan=TIntervalLoops(interval=datetime.timedelta(seconds=4), loops=3) z_plan=ZRangeAround(range=10.0, step=0.3) keep_shutter_open_across=('z',)
Traceback (most recent call last):
  File "c:\Control_2\Zeiss-microscope\control_2\src\zeiss_control\backend\test_useq_save.py", line 34, in <module>
    assert wdg.value().replace(metadata={}) == MDA_noZ
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError
```

tlambert03, 2024-11-06:
fix incoming :joy:

fdrgsp, 2024-11-06:
> fix incoming 😂

🤣👌

PR #381 (2024-11-06): fix: fix include_z when setting stage positions value in PositionTable widget
fixes #380 

for now, this will set the "include z" checkbox to True if _any_ of the positions have a Z position.   If only some of them have a z position, it will emit a warning.

If we need the feature in the future, we could further elaborate to somehow let some rows have a Z position of None while others don't
closed 2024-11-06 by tlambert03