pymmcore-plus/pymmcore-widgets #401: Absolute Bounds Grid

dianewiener, 2025-02-21:
The Absolute Bounds Grid MDA GUI is not working. The Top, Left, Right, and Bottom values are updated in the GUI, but then when the image acquisition is run, the stage moves to (0.0, 0.0) in stage coordinates and collects only at that position, regardless of the grid size. The MDA works for the `Fields of View` and `Width & Height` grid options.

As an example, I selected the following bounds

| Position| Stage Coordinates|
|--------|--------|
| Top| 72473.64 µm|
| Left | 19916.76 µm|
| Right| 108151.91 µm|
| Bottom| 11412.33 µm| 

and this is the Terminal Output:

<details><summary>Terminal Output</summary>
<p>

> [02/20/25 15:49:14] INFO     MDA Started:                        _runner.py:376
>                              metadata={'pymmcore_widgets':
>                              {'version': '0.9.0'},
>                              'napari_micromanager':
>                              {'split_channels': False}}
>                              axis_order=('p', 'g', 'c')
>                              stage_positions=(AbsolutePosition(x
>                              =19916.762263549997,
>                              y=72473.63965129, z=2165.487),)
>                              grid_plan=GridFromEdges(fov_width=5
>                              324.8, fov_height=5324.8, top=0.0,
>                              left=0.0, bottom=0.0, right=0.0)
>                              channels=(DEBUG:zarr.core.sync:Creating Zarr event loop
> ChannelDEBUG:asyncio:Using selector: SelectSelector
> (config='Brightfie
>                              ld', exposure=100.0),)
> INFO:pymmcore-plus:MDA Started: metadata={'pymmcore_widgets': {'version': '0.9.0'}, 'napari_micromanager': {'split_channels': False}} axis_order=('p', 'g', 'c') stage_positions=(AbsolutePosition(x=19916.762263549997, y=72473.63965129, z=2165.487),) grid_plan=GridFromEdges(fov_width=5324.8, fov_height=5324.8, top=0.0, left=0.0, bottom=0.0, right=0.0) channels=(Channel(config='Brightfield', exposure=100.0),)
> DEBUG:napari.layers.base.base:Layer.refresh: Exp_b801de90-a915-4558-97ce-95dba018843b
> DEBUG:napari.layers.base.base:Layer._refresh_sync: Exp_b801de90-a915-4558-97ce-95dba018843b
> DEBUG:napari.components._layer_slicer:_LayerSlicer.submit: layers=[<Image layer 'Exp_b801de90-a915-4558-97ce-95dba018843b' at 0x1989fa87140>], dims=ndim=5 ndisplay=2 order=(0, 1, 2, 3, 4) axis_labels=('0', '1', '2', '3', '4') rollable=(True, True, True, True, True) range=(RangeTuple(start=np.float64(0.0), stop=np.float64(0.0), step=np.float64(1.0)), RangeTuple(start=np.float64(0.0), stop=np.float64(0.0), step=np.float64(1.0)), RangeTuple(start=np.float64(0.0), stop=np.float64(0.0), step=np.float64(1.0)), RangeTuple(start=np.float64(0.0), stop=np.float64(5322.2), step=np.float64(2.6)), RangeTuple(start=np.float64(0.0), stop=np.float64(5322.2), step=np.float64(2.6))) margin_left=(0.0, 0.0, 0.0, 0.0, 0.0) margin_right=(0.0, 0.0, 0.0, 0.0, 0.0) point=(np.float64(0.0), np.float64(0.0), np.float64(0.0), np.float64(0.0), np.float64(0.0)) last_used=0, force=False
> DEBUG:napari.components._layer_slicer:Sync slicing for Exp_b801de90-a915-4558-97ce-95dba018843b
> DEBUG:napari.layers.base.base:Layer._slice_dims: Exp_b801de90-a915-4558-97ce-95dba018843b, dims=ndim=5 ndisplay=2 order=(0, 1, 2, 3, 4) axis_labels=('0', '1', '2', '3', '4') rollable=(True, True, True, True, True) range=(RangeTuple(start=np.float64(0.0), stop=np.float64(0.0), step=np.float64(1.0)), RangeTuple(start=np.float64(0.0), stop=np.float64(0.0), step=np.float64(1.0)), RangeTuple(start=np.float64(0.0), stop=np.float64(0.0), step=np.float64(1.0)), RangeTuple(start=np.float64(0.0), stop=np.float64(5322.2), step=np.float64(2.6)), RangeTuple(start=np.float64(0.0), stop=np.float64(5322.2), step=np.float64(2.6))) margin_left=(0.0, 0.0, 0.0, 0.0, 0.0) margin_right=(0.0, 0.0, 0.0, 0.0, 0.0) point=(np.float64(0.0), np.float64(0.0), np.float64(0.0), np.float64(0.0), np.float64(0.0)) last_used=0, force=False
>                     INFO     index=mappingproxy({<Axis.POSITION: _runner.py:337
>                              'p'>: 0, <Axis.GRID: 'g'>: 0,
>                              <Axis.CHANNEL: 'c'>: 0})
>                              channel=Channel(config='Brightfield
>                              ') exposure=100.0 x_pos=0.0
>                              y_pos=0.0 z_pos=2165.487
> INFO:pymmcore-plus:index=mappingproxy({<Axis.POSITION: 'p'>: 0, <Axis.GRID: 'g'>: 0, <Axis.CHANNEL: 'c'>: 0}) channel=Channel(config='Brightfield') exposure=100.0 x_pos=0.0 y_pos=0.0 z_pos=2165.487
> [02/20/25 15:49:22] INFO     MDA Finished:                       _runner.py:463
>                              metadata={'pymmcore_widgets':
>                              {'version': '0.9.0'},
>                              'napari_micromanager':
>                              {'split_channels': False}}
>                              axis_order=('p', 'g', 'c')
>                              stage_positions=(AbsolutePosition(x
>                              =19916.762263549997,
>                              y=72473.63965129, z=2165.487),)
>                              grid_plan=GridFromEdges(fov_width=5
>                              324.8, fov_height=5324.8, top=0.0,
>                              left=0.0, bottom=0.0, right=0.0)
>                              channels=(Channel(config='Brightfie
>                              ld', exposure=100.0),)
> INFO:pymmcore-plus:MDA Finished: metadata={'pymmcore_widgets': {'version': '0.9.0'}, 'napari_micromanager': {'split_channels': False}} axis_order=('p', 'g', 'c') stage_positions=(AbsolutePosition(x=19916.762263549997, y=72473.63965129, z=2165.487),) grid_plan=GridFromEdges(fov_width=5324.8, fov_height=5324.8, top=0.0, left=0.0, bottom=0.0, right=0.0) channels=(Channel(config='Brightfield', exposure=100.0),)

</p>
</details> 

I think the widget values aren't being updated and passed to the grid_plan sequence.

I am using the following `napari-micromanager = 0.1.3`, `pymmcore-plus = 0.13.3`, `useq-schema = 0.7.0`, `pymmcore-widgets = 0.9.0` on 
Windows 11 Pro.

Happy to help debug as needed!

fdrgsp, 2025-02-21:
Hi @dianewiener, thanks for the issue! I believe we fixed this in #396 but we did not release the updated version of `pymmcore-widget` yet. We'll do that soon!

dianewiener, 2025-02-26:
Thanks @fdrgsp ! Do you know when this and PR #402 will be released? (We ran into the PR #402 issue as well.) An updated version would be helpful for us so that we don't have to work around these issues

tlambert03, 2025-02-26:
sorry for the delayed release!  We just pushed v0.9.1 now, it should be on PyPI shortyl

dianewiener, 2025-02-26:
Thank you! We really appreciate it! It works now!

closed 2025-02-26 by dianewiener