pymmcore-plus/pymmcore-widgets #405: MDA widget grid plan does not use the current ROI

SJShep, 2025-03-05:
We are using pymmcore-gui and loading the MDA widget.

When generating stage positions using the Grid plan, the resulting grid uses the whole camera chip rather than the current ROI size. 
Do you have any hints for how to setup the grid using the current camera ROI or the grid plan takes the requested ROI size?

fdrgsp, 2025-03-05:
Hi @SJShep, thanks for reporting this issue!

It looks like a bug in the widget, likely due to the fact that we haven’t connected the core’s `roiSet` signal to the method responsible for updating the camera ROI size.

We’ll work on fixing this soon!

Just to clarify, do you observe this issue when running an acquisition? The core engine should be setting the camera ROI correctly in that case...

SJShep, 2025-03-05:
Hi @fdrgsp, thanks for the quick response!

We are using a custom acquisition engine which updates the ROI dynamically. We hope to use the grid to generate positions for a hardware sequenced acquisition.