pymmcore-plus/pymmcore-widgets #411: docs: What's with the Pink Presets?

gselzer, 2025-03-11:
After a fair amount of digging into the source code, I can see that pink text [specifies](https://github.com/pymmcore-plus/pymmcore-widgets/blob/37171860471fdf77dcfef466d5a8bb999af5ca72/src/pymmcore_widgets/control/_presets_widget.py#L119) that there is no preset that can capture the current system state, however this was not clear to me (without digging into the code) - this may well be something astute users can pick up on, but not me! (I actually discussed this with @marktsuchida multiple times 😅)

At a minimum, I'd appreciate if this shows up in documentation somewhere. Ideally, the tooltip would explain the situation. Happy to implement any agreed upon changes.

Looking back at mmstudio, if no preset matches the system state, the combo box is cleared, which I think more clearly signifies that no preset applies to this situation. @fdrgsp is there a reason you prefer the pink text? The use of color does help to grab attention to the preset...

Another disadvantage of the cleared combobox in mmstudio is that if you click on a blank preset box, the system state will change when it loses focus.

fdrgsp, 2025-03-12:
Hey @gselzer, when we created the widget, I chose to use a color instead of leaving the field empty as a personal preference. It doesn’t have to be `magenta`; we could use `light gray`, for example, to give it a “disabled” appearance. We can also add to the first line of the tooltip something indicating that the preset is not set.
However, if everybody thinks it is confusing we can think about going back to the empty combo...

tlambert03, 2025-03-12:
it does seem like a number of people have been confused by it.  Color aside: perhaps even just having the "last" commanded setting is too distracting and it should be all or nothing.  So yeah: empty combo works for me

gselzer, 2025-03-12:
> It doesn’t have to be `magenta`; we could use `light gray`, for example, to give it a “disabled” appearance.

Well, I think the color you did choose is quite nice - it has sufficient contrast to grab your attention. I would think a more red color would be nice to convey the meaning of "something's not right here", although that would deviate from the green/pink theme the widgets have.

> However, if everybody thinks it is confusing we can think about going back to the empty combo...

I could certainly attempt a fix. The one UI feature I'd want to make sure we include if we go in that direction is that, if a combobox is empty/magenta, if you click it (just to view the options) and then click off of it, that should not trigger the setting of any preset (as it seems to do in mmstudio)

marktsuchida, 2025-03-13:
> if a combobox is empty/magenta, if you click it (just to view the options) and then click off of it, that should not trigger the setting of any preset

This will be nice indeed.

My vote is for not showing the last matching preset, not just because it may not even be approximately correct in many situations, but also because that information is not easy to obtain correctly (if there is no match when the widget is created, how would it know what the last matching preset was?).

On the other hand, the meaning of the empty box is also not always clear to people, I have seen. It might be nice to show something like `<no match>` in this case, if not too hard (maybe in italics).

gselzer, 2025-03-13:
> It might be nice to show something like `<no match>` in this case, if not too hard

Ooh, I like something along this idea.

gselzer, 2025-07-01:
> (I actually discussed this with [@marktsuchida](https://github.com/marktsuchida) multiple times 😅)

One of those times probably resulted in the creation of #325. Going to mark that one as a duplicate, since I think this has more relevant conversation.

gselzer, 2025-07-02:
Yup, definitely interested in tackling this soon

PR #466 (2025-08-21): fix: remove magenta from preset widget and use <no match> key
This PR implements the <no match> logic discussed in #411.

BEFORE
[image]

AFTER
[image]

closes #411.