pymmcore-plus/pymmcore-widgets #450: Incorrect images in docs widget examples

gselzer, 2025-07-01:
Looks like the wrong image is generated for at least the [`StageWidget`](https://pymmcore-plus.github.io/pymmcore-widgets/widgets/StageWidget/) and the [`ImagePreview`](https://pymmcore-plus.github.io/pymmcore-widgets/widgets/ImagePreview/) docs pages...

tlambert03, 2025-07-01:
thanks... yeah, there can be a race condition that sometimes prevents the right top level widget being grabbed (i.e. it may grab one from a previous run).  

there is code in this branch that always picks the right one by using a subprocess for every widget.  Slower, but always right:

https://github.com/pymmcore-plus/pymmcore-widgets/blob/7c5343011f30b6aa9bc72ae48470ebc3b01d2eb3/docs/_hooks.py#L140-L176

we could use that in main

gselzer, 2025-07-02:
> Slower, but always right

Gotcha...like how much slower? It'd be great if we could guarantee correctness, at least for pushes to `main`?

tlambert03, 2025-07-02:
not so slow we shouldn't use it on main as I implied above :)